YorffoeG/diyscip #2: support for intex Saltwater Spa Hexagon 28456

red-gee, 2020-06-11:
would It be possible to get support for the 28456 Saltwater Spa? 

if you need some pictures of the control-panel PCB, I can help.

YorffoeG, 2020-06-12:
Maybe it's already works with a 28456 but without jet and cleaning control/command.
I plan to support it in the next weeks, need first to find this model to test.

dvoc, 2020-06-17:
Great project! Interested too in controlling 28456 Saltwater Spa and integration with Home Assistant (mqtt mosquitto server).
Additionally I need to know if it's possible to use this analog mux from amazon instead of  CD4051BE. 

https://www.amazon.es/Ils-74HC4051-Canales-multiplexor-Anal%C3%B3gico/dp/B07912SZTD/ref=sr_1_1?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=74hc4051&qid=1592393806&s=electronics&sr=1-1

BR

YorffoeG, 2020-06-17:
Maybe 74HC4051 should works... it's 5V only compare to cd4051, but signal INH, A, B, C are 3.3V... enough to enable address and output of multiplexer... probably but never tried.

red-gee, 2020-06-17:
Hi BR, 

Integrating it in Homeassistant seems to be no problem, I run the origin version from Geoffroy and integrated it in ioBroker via mqtt. 

I will Test the 74hc on breadboard next days and will tell you about the results. 
Mit freundlichen Grüßen/ regards 

Ralf

dvoc, 2020-06-17:
Hi red-gee,
Many thanks for your help.
Are you able to control jets and clean system by the control panel of the spa after installing the circuit?

red-gee, 2020-06-17:
@dvoc my setup till now is: only a naked node-mcu on the spa-control-panel, just to test it. Till now I can control the panel without any problems. I will try the setup with shifting-register and muxer the next days! As soon as I get results I will post it here! 

Regards

YorffoeG, 2020-08-05:
SJB model (28456) now supported.

closed 2020-08-05 by YorffoeG

jperquin, 2020-09-02:
This is working flawlessly and very stable, so very happy!

My setup is as follows:

- Geoffroy's ready made board installed on the 28456 (with salt water cell).
- Home Assistant with MQTT add-on (as MQTT broker) which can be monitored via MQTT Explorer for debugging.
- The board connects via WIFI to the MQTT broker / HA allows for seamless interaction with the MQTT broker.
[image]

in sensors.yaml:

`#Spa Controller
- platform: mqtt
  state_topic: "spa/model"
  name: "Spa Model"
- platform: mqtt
  state_topic: "spa/status"
  name: "Spa Status"
- platform: mqtt
  state_topic: "spa/sys/version"
  name: "Spa Firmware version"
- platform: mqtt
  state_topic: "spa/sys/wifi"
  name: "Spa Wifi Signal"
- platform: mqtt
  state_topic: "spa/state"
  name: "Spa State Debug"
- platform: mqtt
  state_topic: "spa/state/heatreached"
  name: "Spa Heat Reached"
- platform: mqtt
  state_topic: "spa/state/bubble"
  name: "Spa Bubble State"
- platform: mqtt
  state_topic: "spa/temp/board"
  name: "Spa Temp Board"
  unit_of_measurement: ºC
- platform: mqtt
  state_topic: "spa/temp/water"
  name: "Spa Temp Water"
  unit_of_measurement: ºC
- platform: mqtt
  state_topic: "spa/temp/desired"
  name: "Spa Temp Target"
  unit_of_measurement: ºC
- platform: mqtt
  state_topic: "spa/sanitizer"
  name: "Spa Sanitizer Timer"`

in switches.yaml
`#Spa Switches
- platform: mqtt
  unique_id: spa_power
  name: "Spa Power"
  state_topic: "spa/state/power"
  command_topic: "spa/state/power/set"
  payload_on: "on"
  payload_off: "off"
  state_on: "on"
  state_off: "off"
  optimistic: false
  qos: 0
  retain: true
- platform: mqtt
  unique_id: spa_filter
  name: "Spa Filter"
  state_topic: "spa/state/filter"
  command_topic: "spa/state/filter/set"
  payload_on: "on"
  payload_off: "off"
  state_on: "on"
  state_off: "off"
  optimistic: false
  qos: 0
  retain: true
- platform: mqtt
  unique_id: spa_heater
  name: "Spa Heater"
  state_topic: "spa/state/heater"
  command_topic: "spa/state/heater/set"
  payload_on: "on"
  payload_off: "off"
  state_on: "on"
  state_off: "off"
  optimistic: false
  qos: 0
  retain: true
- platform: mqtt
  unique_id: spa_jets
  name: "Spa Jets"
  state_topic: "spa/state/jet"
  command_topic: "spa/state/jet/set"
  payload_on: "on"
  payload_off: "off"
  state_on: "on"
  state_off: "off"
  optimistic: false
  qos: 0
  retain: true
- platform: mqtt
  unique_id: spa_bubble
  name: "Spa Bubbles"
  state_topic: "spa/state/bubble"
  command_topic: "spa/state/bubble/set"
  payload_on: "on"
  payload_off: "off"
  state_on: "on"
  state_off: "off"
  optimistic: false
  qos: 0
  retain: true`

I am also running a couple of automations to save electricity cost:
- run the filter only for 15 mins every hour
- only run spa heater on lower electricity tariff (saves about 10%)

ashmodai6, 2020-09-03:
@jperquin your slider to set the desired temp looks nice. Is it a custom component ?

jperquin, 2020-09-04:
@ashmodai6 this slider for setting temperature requires adding the following:

Under HELPERS add:
[image]

and in automations.yaml:
```
- id: set_spa_temp_slider
  alias: Set temp slider
  trigger:
  - entity_id: sensor.spa_temp_target
    platform: state
  action:
  - data_template:
      entity_id: input_number.spa_temp_target
      value: '{{ trigger.payload }}'
    service: input_number.set_value
  mode: single
- id: spa_temp_slider_moved
  alias: Temp slider moved
  trigger:
  - entity_id: input_number.spa_target_temp
    platform: state
  action:
    service: mqtt.publish
    data_template:
      topic: spa/temp/desired/set
      retain: true
      payload: '{{ states.input_number.spa_target_temp.state | int }}'
```

ashmodai6, 2020-09-04:
@jperquin Thanks a lot !

giloris1, 2020-09-07:
@jperquin 
Hi thank you for yours explains. I have the same projet , i  bought module to Geoffroy.
Every sensors and switchs runs very well , but i have problem to automation with slider to have temp heater changed
When i move slider , spa temp desired not change, When i use developpement tools service, and write commands or on mqtt directly spa temp desired change !

Can you give me link to HELPERS add to configure slider.
 
My config is :
/config/automations.yaml

- id: '1599488226573'
  alias: set_spa_temp_slider
  description: ''
  trigger:
  - entity_id: sensor.mqtt_sensor_2
    platform: state
  condition: []
  action: 
  - data_template:
      value: '{{ trigger.payload }}'
      entity_id: input_number.target_temp
    service: input_number.set_value
  
   
- id: '1599488415070'
  alias: spa_temp_slider_moved
  description: ''
  trigger:
  - entity_id: input_number.target_temp
    platform: state
  condition: []
  action:
    service: mqtt.publish
    data_template:
      payload: '{{state.input_number.target_temp.state | int }}'
      retain: true
      topic: spa/temp/desired/set
      service: mqtt.publish

Thank to help me.
Giloris

jperquin, 2020-09-11:
HELPERS are under Configuration in HA