YorffoeG/diyscip #15: Desired temp not published as retained message

MartnIV, 2020-10-26:
Hey,

Desired temp not published as retained message, this blocks some of my workflows.
I've made a led indicator in the house to see if the desired goal is reached but that workflow would need to know the desired temps all the time.

Is it something you could change?

YorffoeG, 2020-10-26:
Hi,

Yes desired temp is published with retained message (like any other message).

But at startup, controller doesn't know the desired temperature until it is set. It's the same on your spa, you don't know the temp until you set it. Maybe you can add in your workflow a desired temperature command when you don't have this information ?

NicoLPHA, 2021-01-16:
Do you sel the pcb??

Nico

lioelec, 2021-05-25:
Salut
Je serai intéressé par l'achat d'une interface fonctionnelle ou à construire.
Disposes-tu du pcb V2 ?

cdlt