YorffoeG/diyscip #17: Woher die Platine

hardlog, 2021-04-17:
Hallo Cooles Projekt,

ich habe eine Intex pureSpa  Model SB-H20 (von 2020), bei dem wird das Display ausgeklappt und hat keine Teleskop-Funktion?
Ich weiß nicht genau wie weit das mit den Modellen deines Projektes kompatibel ist? Das Display sieht zumindest ähnlich aus...

Geht das auch mit dem Gerät?

Woher kann ich deine Steuerplatine (im Idealfall bestückt) bekommen?

Gruß
hardlog

rhaamo, 2021-04-22:
considering an SB-H20 too but couldn't find anything, I guess it also uses the same 5pin connector but no idea.

@YorffoeG any ideas ?

rhaamo, 2021-04-22:
According to https://github.com/jnsbyr/esp8266-intexsbh20 the protocols between the SB-H20 and the ones handled by this project aren't compatible, you should look at the other one if you plan getting an SB-H20

YorffoeG, 2021-04-22:
Hi, some users use the project with 2020 model SB-H20, it works but not with the model with remote control panel.

rhaamo, 2021-04-22:
ok that's weird since another projects says that the SB-H20 protocol is different.
the SB-H20 has the display built into the whole controller, and the ones with the deported screen higher don't seems in stock everywhere and the wifi ones are both different models.

I guess I will get an SB-H20 and see with the unit in hands.

baptistevenandy, 2022-01-21:
Hello all, i have some extra pcb (min 5 to order) with components (to solder). I am testing it right now. You can ask me if you are interesting.

Jamster0815, 2022-08-13:
> Hallo Cooles Projekt,
>
> ich habe eine Intex pureSpa Model SB-H20 (von 2020), bei dem wird das Display ausgeklappt und hat keine Teleskop-Funktion? Ich weiß nicht genau wie weit das mit den Modellen [...]

Hy !

hast du eine Antwort bezüglich der Gerber Files bzw der Platine bekommen ??

Danke BG Christoph