YorffoeG/diyscip #19: Code erreur E90?

JeedHome44, 2021-09-12:
Bonjour,

la carte marche bien mais j’aimerai savoir ce qui est envoyé par MQTT quand le spa se met en erreur E90?

j’aimerai récupérer cette erreur sur Jeedom pour me notifier de changer le filtre

merci

jerryzz, 2021-09-12:
tu as le code d'erreur sur l'info IntexSpa/Error Number

JeedHome44, 2021-09-12:
Oui je sais. Quand le spa s'est mis en E90, j'ai regardé le code erreur et je l'ai mis en déclencheur d'un scénario Jeedom mais il ne se reproduit plus. Ce n'est donc pas le bon

jerryzz, 2021-09-12:
ca fonctionne très bien sur déclencheur "Action sur la valeur"

[image]

JeedHome44, 2021-09-12:
Ah mais t'as pas du comprendre ma question.

Je ne veux pas recevoir tous les codes erreurs.

Je veux juste savoir quand le spa se met en mode E90.

Ça signifie qu'il faut changer le filtre sale.

jerryzz, 2021-09-12:
Ben tu met "égal 90" à la place de "différent 0" ... ?!?

roudou, 2021-09-12:
C'est ce que j'ai fait et à chaque fois je reçois un telegram qui me dit
change le filtre du spa ;)

JeedHome44, 2021-09-12:
A donc vous êtes en train de dire que n'importe quel code erreur de la carte signifie que c'est un code E90 sur le spa?

Le spa a d'autres codes erreur normalement non?

roudou, 2021-09-12:
Je connais pas les autres code (et cela me va bien ;)  j'ai juste mis si
status contient 90 envoie un message télégramme ;) avec en déclencheur le
status qui change

JeedHome44, 2021-09-12:
Ah ok!

L'erreur E90 alors doit s'afficher "90" sur MQTT?

Je viens de changer le scénario pour me notifier chaque erreur.

Je vais pouvoir identifier celle qui correspond au E90.

closed 2022-12-21 by JeedHome44